mazi-project/guides #2: documentation on demo server outdated

panosnethood, 2017-05-02:
The documentation on the demo server is outdated: http://demo.mazizone.eu:4567/index_documentation. 

Perhaps create a script to keep the documentation up-to-date based on the github wiki, both on the demo server and on the raspberry Pi when a users does an update?

haniavis, 2017-05-03:
The documentation is updated based on the github wiki. 

Please notice that one part of the info [here](https://github.com/mazi-project/guides/wiki/Applications) is for the administrator and another part for the user. So the administrator tips are included in the [documentation of the admin Portal](http://demo.mazizone.eu:4567/admin_documentation).

panosnethood, 2017-05-16:
Both in the demo server and version 1.6.5 the user documentation is not updated according the latest version of the github wiki here: https://github.com/mazi-project/guides/wiki/Applications (it would be great if the update is somehow automatic)

closed 2017-09-26 by haniavis