mazi-project/guides #3: Guidelines for typical Raspberry Pi accessories (USB storage, display)

panosnethood, 2017-11-22:
Hello,

I would like to experiment with two accessories for a Raspberry Pi and I would appreciate some help.

First, I would like to add an external USB hard disc and connect it to the NextCloud application and others perhaps. Is this possible? It seems easy.

Second, I have bought this touchscreen display, https://www.raspberrypi.org/products/raspberry-pi-touch-display/, and I would like to enable the "Desktop" interface to use it in a MAZI Zone installation. I imagine that this is not included in the standard MAZI image. Is there an easy way to install it afterwards or perhaps it would be better if a 2nd version of the image with the GUI installed was made available?

Thanks!

mgaved, 2017-11-29:
The OU has purchased a 5" screen (http://www.mouser.com/ds/2/737/adafruit-5-800x480-tft-hdmi-monitor-touchscreen-ba-932828.pdf) . There is a guide for editing the /boot/config.txt file to enable the display to work (force video resolution), but so far I've only used it to view the start up code to check progress on booting up, I've not explored whether the MAZItoolkit includes a GUI (xwindows or similar). Would there be an issue with this taking up much more space on the installation? 

Configuring the Adafruit 5" and 7" 800x480 TFT HDMI Backpack to work with Raspberry Pi (tested as working for MAZI toolkit v2.0 and v2.1)
https://learn.adafruit.com/adafruit-5-800x480-tft-hdmi-monitor-touchscreen-backpack/raspberry-pi-config

panosnethood, 2017-11-30:
Thanks Mark! What is nice about the display I linked above is that it does not need any battery, since it takes power from the Raspberry Pi. The problem then is that the whole thing is a little more difficult to manage and for this it would be nice if we could create a nice "case" or even better a "box" to put on the wall.

I will post some photos on this soon ...

ingiH, 2017-11-30:
Hi Panos and Mark,

I was looking at these touchscreens recently and there are cases available:
https://thepihut.com/products/raspberry-pi-official-7-touchscreen-case

Ingi

…………………………………………...…………………………………………….

Ingi Helgason, Centre for Interaction Design, Edinburgh Napier University
MAZI: DIY networking (EU CAPS H2020)
www.ingihelgason.co.uk<http://www.ingihelgason.co.uk>

This message and its attachment(s) are intended for the addressee(s) only and should not be read, copied, disclosed, forwarded or relied upon by any person other than the intended addressee(s) without the permission of the sender. If you are not the intended addressee you must not take any action based on this message and its attachment(s) nor must you copy or show them to anyone. Please respond to the sender and ensure that this message and its attachment(s) are deleted.

It is your responsibility to ensure that this message and its attachment(s) are scanned for viruses or other defects. Edinburgh Napier University does not accept liability for any loss or damage which may result from this message or its attachment(s), or for errors or omissions arising after it was sent. Email is not a secure medium. Emails entering Edinburgh Napier University's system are subject to routine monitoring and filtering by Edinburgh Napier University.

Edinburgh Napier University is a registered Scottish charity. Registration number SC018373

mgaved, 2017-11-30:
Very cool, Panos. I think my screen should also potentially be able to run from the Raspberry PI, I have to investigate the power settings. But at the moment I have one of the 10,000mAh EasyAcvc batteries, with one cable to the Pi and one ot the monitor, so I guess it's the same :-) 

I agree on the case, let me know if you come across any good CAD models, I can get one laser cut in wood here. I've seen some nice ideas in armoured Peli Cases but these are quite expensive boxes. 

For a development machine, and for outdoor use, I am also interested in 'rugged connectors' - I am a bit worried that the constant plugging in and out of power and monitor cables might end up damaging the connectors on the motherboard of the PI, so this something I would like to investigate. I wonder if people have done work on this? There are all sorts of stronger connectors, e.g. https://www.solwise.co.uk/networking_sundries_ip67-socket.htm . A slightly larger case might allow short cable runs (500-100mm?) from the onboard connectors to stronger connectors on a case to protect the Pi?

panosnethood, 2017-11-30:
Thanks Ingi! And yes Mark, the Raspberry Pi can offer power to USB-powered devices through its USB ports. I have tried with the MR-3020 router and it works.

deckspace, 2017-12-27:
External hard drives even SSD models will draw too much power from the PI.
Poor or under powered Raspberry Pi could damage the micro SD card and corrupt your data. The Raspberry PI requires a 2.1 amp 5V PSU or battery despite drawing only 0.5amp on average. Lower powered or shared PSU are not advisable certainly if you are adding Screens or SenseHAT  .