mazi-project/guides #4: Explain properly the two different types of wireless connectivity

panosnethood, 2017-11-30:
The Raspberry Pi has an internal antenna that is used to accept incoming connections, it acts as an Access Point.

The additional USB dongle can then be used to connect the Raspberry Pi to a close by Internet source, i.e., a wireless router at home or elsewhere, if for example one cannot connect it directly to the Internet router with an ethernet cable.

This is a little complicated for non-technically savvy people and perhaps a nice figure to explain this should be made.

I will wait in case there any volunteers for this and otherwise I will give it a try myself :-)

haniavis, 2017-12-12:
Hi Panos,

Did you have any progress on that?

If not, I can give it a try.

panosnethood, 2017-12-12:
Not yet, and I don't think I will have the time to do something before the end of the year.

deckspace, 2017-12-27:
The integrated wireless network adapter (wlan0) of the Raspberry Pi (3) is used in access point mode (AP).

An additional USB wireless network adapter can be configured for connection to a wlan router as alternative to an ethernet cable.

Other combined uses of ethernet and wlan adapters are also possible when extending wlan coverage over a larger area.