mazi-project/guides #5: Administration panel access

zku-berlin, 2017-12-07:
Dear all,

I would like to access the administration panel but always end up on the demo mode http://demo.mazizone.eu:4567/admin_network

Do I need to be connected to internet and to the Mazi to access it?
Also, when I reset the password did I also reset the user name? And if not, is it possible to reset it?

Thank you for your answers,

Charles

panosnethood, 2017-12-07:
Hi Charles,

No, you don't have to be connected to the Internet to access the admin page. But make sure that you are connected to your MAZI Zone through WiFi, and then try http://local.mazizone.eu/admin/.

This URL always works, and then at the menu "Networking" you can check which is the "custom" URL for your MAZI Zone. In the default installation the custom URL is portal.mazizone.eu but you might have changed it already?

As for the resetting, you only reset the password to "1234". The web admin's username stays always "mazi_admin". Would you like to be able to change it? If yes, you create a new issue here: https://github.com/mazi-project/portal/issues

Hope this helps.

zku-berlin, 2017-12-07:
Hi again !

It helped in deed, thanks ! Now I can log in to the admin panel however, I had to type http://portal.mazizone.eu/admin/ and not http://local.mazizone.eu/admin/ to access it as a custom URL, it might be that I changed it when updating the Mazi...

As for the user name, I'me happy as long as it's working but thank you for asking.

Problem solved ! Thanks again for your help !

panosnethood, 2017-12-07:
Nice! Normally, http://local.mazizone.eu/ should also work (this cannot be changed by the admin), not sure what is wrong but if it works now you shouldn't bother.

haniavis, 2017-12-12:
Hi Charles and Panos,

You can change both the username and the password of the admin panel, through the top right menu in the admin panel. 

The URL local.mazizone.eu/admin is not supposed to work. Instead you can use either portal.mazizone.eu/admin (this depends on the URL you have defined in the admin panel) or the local.mazizone.eu:4567/local (this should always work)

Best regards,
Harris

panosnethood, 2017-12-12:
Thanks Harris. Sorry for the confusion, I think that I could access through http://local.mazizone.eu/admin ... I will check again. And in any case I will add this detail in the guidelines.

deckspace, 2017-12-27:
http://local.mazizone.eu:4567/admin will always redirect to http://local.mazizone.eu:4567/admin_dashboard

If it doesn't you need to clear your browser cache.

closed 2018-01-30 by haniavis