mazi-project/guides #8: etherpad plugins / nextcloud applications

panosnethood, 2017-12-29:
It would be nice to include in the guidelines a review of existing etherpad plugins and optional nextcloud applications (and also installation tips if needed) and perhaps pre-install those that we found useful and stable.

For example, there is an etherpad plugin for adding comments that if it works well, would be very nice to have: https://github.com/ether/ep_comments

(please add below recommendations for other plugins and/or applications)

haniavis, 2018-03-08:
Hi,

The following have been integrated in v2.4:
- Etherpad plugin for commenting, ep_comments_page + ep_page_view
- Nextcloud app for mounting external storage

In case there are more useful plugins to be added, please mention them here.

mgaved, 2018-11-17:
Automatically listing all the etherpads created would be ideal for some of the OU scenarios. I'll try ep_pad-lister , though it has indicated the following error:

pi@mazizone:/var/www/html/etherpad-lite $ sudo npm install ep_pad-lister
npm WARN enoent ENOENT: no such file or directory, open '/var/www/html/etherpad-lite/package.json'
npm ERR! Linux 4.9.31-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "ep_pad-lister"
npm ERR! node v6.7.0
npm ERR! npm  v3.10.3
npm ERR! path /var/www/html/etherpad-lite/node_modules/ep_comments_page
npm ERR! code EISGIT
 
npm ERR! git /var/www/html/etherpad-lite/node_modules/ep_comments_page: Appears to be a git repo or submodule.
npm ERR! git     /var/www/html/etherpad-lite/node_modules/ep_comments_page
npm ERR! git Refusing to remove it. Update manually,
npm ERR! git or move it out of the way first.
 
npm ERR! Please include the following file with any support request:
npm ERR!     /var/www/html/etherpad-lite/npm-debug.log
 
 
 
Many thanks!
Mark