mazi-project/guides #9: How to Contribute Documentation?

hellekin, 2018-02-06:
Hi,

I'm trying to solve #7, but I'm confused with the source tree, as it seems to contain build files, and has 2 source directories, one called `_sources` containing text files, and one called `source` containing `.rst` files.

- How are these files and folders related?
- What is the procedure to change text or add new files?
- How to build the documentation locally to verify its integrity before pushing PRs?

Thank you!

panosnethood, 2018-02-06:
Thanks @hellekin! Looking at previous commits, it seems that it should be enough to change the .rst files. @haniavis could help on how you can build locally the documentation, but for simple changes you could already try a pull request directly changing the corresponding .rst file.

panosnethood, 2018-02-07:
I updated here the guidelines for contributors: https://github.com/mazi-project/guides/wiki/Contributing

Next step is to write guidelines for the contributors to the software code ...

hellekin, 2018-02-07:
OK. I think there should be a Makefile to build the documentation, and the `build/` directory should not be in the repository, only the sources. That would remove confusion... Maybe a [Dockerfile](https://hub.docker.com/r/dldl/sphinx-server/) could be there as well for people to preview the build without polluting their system with Sphinx dependencies.

hellekin, 2018-02-07:
Hi @haniavis, I made #10 that should also close this issue.

PR #10 (2018-02-07): Fix documentation (closes #7)
Hello,

this PR introduces a `Makefile`, removes the generated files (in `tech/build`), adds instructions to build the HMTL documentation using `sphinx-build` (closes #9), and -- its original intent -- fixes the progress information on the `dd` command, effectively closing issue #7.
closed 2018-02-07 by haniavis