mazi-project/guides #14: password for nextcloud and interview admin page

AlexandrosPapag, 2018-02-19:
Hi, 

I am Alexandros, I've been using the mazizone for a while, but I am still exploring it. I am not an expert on technology, quite the opposite I would say. The other day I was trying to access the nextcloud app and I also wanted to edit an interview I had taken. I realized that there's a different password needed than the 'main' one. I looked for it, but couldn't find it. 
Could this information be placed in a more obvious position, somewhere that even non computer-savvy users would come upon very easily?
Thank you,
Alexandros

haniavis, 2018-02-22:
Hi Alexandros,

the default admin credentials for all the applications in the MAZI toolkit are:
admin
password

and can be found [here](http://nitlab.inf.uth.gr/mazi-guides/download.html) at the bottom of the page.

AlexandrosPapag, 2018-02-22:
Thanks Hari!

haniavis, 2018-02-23:
I apologize for the wrong default credentials!

The correct ones are:
admin
mazizone

panosnethood, 2018-02-23:
Thanks Alexandros! Yes, we need to put this information in a more visible place. Perhaps it would make sense to make a small MAZI admin "cheat sheet" or something like this and even print it to hand it over in workshops we organize.

panosnethood, 2018-06-21:
Here is a first version of the "fact sheet": https://github.com/mazi-project/guides/wiki/Administrator-cheat-sheet

Perhaps we could integrate it in the admin panel documentation as a nice pdf to be printed (with the option to change some fields like the custom URL, and the passwords?

P.S. I realize that the whole set-up with the different URLs is rather complicated and we could try to simplify it (like not having a different port for the admin URL in the two URL options (local.mazizone.eu & custom URL).