mazi-project/guides #17: TL-MR3020 V3 – no OpenWRT Support?

mariusfoerster, 2018-10-25:
Hi! I want to use the tp link TL-MR3020 as an external open wrt router. Unfortunately, after buying one, I found out that the newest version TL-MR3020 V3 is not supported by Open WRT. I found some information but wasn't able to understand it. 

Do you have an Idea how to fix this problem? 

Furthermore, it would be very helpful for others to know, that there are different versions of the TL-MR3020 available and that this could cause some troubles...

THX!

panosnethood, 2018-10-25:
HI Marius, you are right! Sorry for this. There are a lot of changes lately regarding OpenWRT compatibility especially for new routers and we need to update the guidelines to make sure that people will not buy the wrong equipment.

For now, one could check here: https://openwrt.org/toh/views/toh_available_864

panosnethood, 2018-10-25:
I have updated the wiki page (added a warning): https://github.com/mazi-project/guides/wiki/External-OpenWRT-router 

See also this issue: https://github.com/mazi-project/portal/issues/129

mariusfoerster, 2018-10-25:
> See also this issue: mazi-project/portal#129 <https://github.com/mazi-project/portal/issues/129>

Thank you very much for the quick answer! Too bad, but great that you added a warning… I’m looking forward to read some updates about possibilities to use routers in the future.

best wishes
m

mariusfoerster, 2018-11-04:
Hey guys,

I have another Issue with my Mazi Zone: I can’t login to the Admin Panel. And I am very sure about the PW. Than I setup a whole new Mazi Zone (Etcher to put the image on the SD card...) but the Problem stayed the same….

sorry for that stupid question….
best
Marius

deckspace, 2018-11-11:
Hi Marius, If you are working with a newly flashed MAZI the first thing that you should see after joining the mazizone wireless network and opening your browser is the initial password setup steps to update the password to one of your choosing (default password is 1234), set the time (Raspberry Pi doesn't have a clock) optional Identification details and setting the wireless network name. If you have completed this step you will have set the mazi_admin password.

panosnethood, 2018-11-11:
See also in these guidelines how to reset the admin password from the command line: https://github.com/mazi-project/guides/wiki/Troubleshooting