mazi-project/guides #20: setup initial admin password does not work

denmarkcontrevida2017, 2019-03-07:
when setting up admin access ,it does not work.

dalelicious, 2019-03-07:
Follow up question. What is the default username for the admin access?
We already tried using "admin" as the username but unfortunately it doesn't work.

panosnethood, 2019-03-07:
Hi, the default username is "mazi_admin" and password "1234".

dalelicious, 2019-03-07:
Thanks a lot! Just saw the updated documentation.

closed 2019-03-07 by panosnethood

deckspace, 2019-03-07:
Well it's not a problem we have experienced. During the setup process the
initial page requests entry of existing password 1234 and your chosen
password.. What happened next?

deckspace, 2019-03-07:
The default username is mazi_admin and password is the one you set

dalelicious, 2019-03-07:
Yeah. It's just we didn't see the updated documentation, but it's fine now. Thanks