mazi-project/guides #22: Raspberry Pi 4 support ?

kermoshina, 2020-07-14:
Hello dear Mazi team! I've been at one of your workshops at Rights Con a few years ago and had a chance to install Mazi on a Raspberry Pi. This time I wanted to try it on a Raspberry Pi 4, but seems that the Pi does not recognize the image. Do you know of anyone who had a successful experience with installing and running a Mazizone on a RasPi 4? If so, please connect me with that person, as I would like to use Mazi for a cool workshop in a forest. Looking forward! Ksenia

panosnethood, 2020-07-21:
Thanks @kermoshina for your interest in MAZI! Indeed the current version is not compatible with the RasPi 4 and the funding for our project has stopped since more than 2 years. Some of us are searching for funding to maintain the platform and/or do some basic adjustments needed with voluntary work, and we will keep you updated when we will succeed :-)

paraguayanwood, 2020-09-12:
Recently i bought a Raspi 4 and wanted to install the mazi zone via the toolkit, i thought i was doing something wrong! Thanks for the explanation @panosnethood, if we could help with something please let us know! Hope that amazing project never stop!